hifiberry/hifiberry-dsp #3: DSPVolume in Alsamixer not working

max2play, 2018-08-13:
Initializing Sigma-Server gives this: 

```
root@max2play:/home/pi# /usr/local/bin/sigmatcpserver --alsa
INFO:root:running as root, data will be stored in /var/lib/hifiberry
INFO:root:initializiong ALSA mixer control
No state is present for card ALSA
Found hardware: "bcm2835_alsa" "Broadcom Mixer" "" "" ""
Hardware is initialized using a generic method
No state is present for card ALSA
ERROR:root:can't create mixer named %s
ERROR:root:ALSA mixer DSPVolume not found
ERROR:root:ALSA mixer not available, aborting volume synchronisation
INFO:root:Announcing via zeroconf
INFO:root:Starting TCP server
```

The Alsamixer element DSPVolume exists but does not have any impact on volume. The setting of DSP-Profiles also doesn't seem to work.

Used default installer, spi enabled, dtoverlay=hifiberry-dac

Thanks!

open-homeautomation, 2018-08-13:
What profile did you install?

hifiberry, 2018-08-13:
What profile did you install?

max2play, 2018-08-14:
tried your profiles
`dsptoolkit install-profile fullrange-mixed.xml`
and
`dsptoolkit install-profile basic-fullrange.xml`

and after a restart the sigmatcp now gives no ALSA error anymore. But the output volume stays the same - there seems to be no connection between the ALSA mixer DSPVolume and the real output volume.

hifiberry, 2018-08-14:
Usually this means there is no profile installed. The profile files exist? What's the output of the install-profile command?

closed 2018-09-18 by hifiberry