hifiberry/hifiberry-dsp #8: DSPToolkit.set_filters off by one overwrites memory

jlindblad, 2019-11-17:
DSPToolkit.clear_iir_filters() calls DSPToolkit.set_filters() with `cutoff_long=True`. I have a custom profile with only six filters, so length is 30 which means `maxlen` is 6. But the loop writing the data to memory will loop 7 times, causing data corruption:

```
        i = 0
        for f in filters:
            logging.debug(f)
            if mode == MODE_LEFT or mode == MODE_BOTH:
                if i < length_left:
                    self.sigmatcp.write_biquad(addr_left + i * 5, f)
            if mode == MODE_RIGHT or mode == MODE_BOTH:
                if i < length_right:
                    self.sigmatcp.write_biquad(addr_right + i * 5, f)
            i += 1
            if i > maxlen:
                break
```

Changing `if i > maxlen` to `if i >= maxlen` fixes the problem.

hifiberry, 2019-11-17:
Thanks for reporting this. I just fixed this in the master branch.

hifiberry, 2019-11-21:
New release 0.18 with this bugfix included has been uploaded to pypi.

closed 2019-11-21 by hifiberry