hifiberry/hifiberry-dsp #9: [ BUG ] AttributeError: 'int' object has no attribute 'readfilters' 0.18

corrpel, 2019-11-25:
Hi,
On 0.18 i receive these error when trying to apply filters:
Traceback (most recent call last):
  File "/usr/local/bin/dsptoolkit", line 29, in <module>
    cmdline.main()
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 937, in main
    self.command_map[cmd]()
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 579, in cmd_set_rew_filters_both
    self.set_iir_filters(mode=MODE_BOTH, format=REW)
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 557, in set_iir_filters
    filters = REW.readfilters(self.args.parameters[0],
AttributeError: 'int' object has no attribute 'readfilters'

hifiberry, 2019-11-25:
What is the command line you're using?

corrpel, 2019-11-25:
dsptoolkit apply-rew-filters default.txt
content of default.txt:

Filter Settings file

Room EQ V5.19
Dated: Nov 18, 2019 7:41:49 PM

Notes:

Equaliser: Generic
No measurement
Filter  1: ON  PK       Fc     80 Hz  Gain  -1.0 dB  Q 0.800
Filter  2: ON  PK       Fc     1000 Hz  Gain  0.5 dB  Q 0.800
Filter  3: ON  PK       Fc     2000 Hz  Gain  0.6 dB  Q 0.800
Filter  4: ON  LS       Fc     60 Hz  Gain   3.0 dB
Filter  5: ON  None
Filter  6: ON  None
Filter  7: ON  None
Filter  8: ON  None
Filter  9: ON  None
Filter 10: ON  None
Filter 11: ON  None
Filter 12: ON  None
Filter 13: ON  None
Filter 14: ON  None
Filter 15: ON  None
Filter 16: ON  None
Filter 17: ON  None
Filter 18: ON  None
Filter 19: ON  None
Filter 20: ON  None

I've downgraded to 0.17 and i don't have this issue.

hifiberry, 2019-11-25:
Ok, I can confirm this. I'll push a bugfix to Github. If you're using pypi, please wait for the 0.19 release. As I'm currently working on some features on the 0.19 release, this will take some time.

corrpel, 2019-11-25:
nice, ok.

SinSilla, 2019-11-25:
Hello,

since i ran into the same problem, can you please tell me how to downgrade to 0.17? Thanks.

hifiberry, 2019-11-26:
I pushed a new 0.19 rlease to pypi that should fix this. Try upgrading to this.

SinSilla, 2019-11-27:
I just tested 0.19 from PyPi and it is working now. Thanks!

hifiberry, 2019-11-28:
Thank you for the feedback

closed 2019-11-28 by hifiberry