hifiberry/hifiberry-dsp #15: Missing metadata readSPDIFOnRegister in dsp profiles

thomasjonas, 2020-05-16:
I was trying to get my LG7450PLA with SoundSync to work with my DAC+DSP. It was quite fun to debug the problem (learning a bunch of new stuff!) and I think it's related to a variable/attribute/metadata not being set in the dsp profiles. In the sigmatcp logs I found the following:

> LG Sound Sync: Using volume register at 4548 and SPDIF active register at None

Which means `ATTRIBUTE_SPDIF_ACTIVE`  or `readSPDIFOnRegister` is undefined. I couldn't find `readSPDIFOnRegister` in any of the DSP profiles xml files, so I guess that must be it. I disabled checking for this register in `soundsync.py` and everything worked (well, after re-uploading my DSP profile, somehow the volume didn't change when the ALSA mixer changed...)

hifiberry, 2020-05-16:
This attribute is used in our latest DSP profile included in HiFiBerryOS:
https://github.com/hifiberry/hifiberry-os/blob/master/buildroot/package/dspprofiles/dspdac-10.xml

It shows that the SPDIF input is active. If you want to create a change make sure that attribute is used if it exists and you may ignore it otherwise.

thomasjonas, 2020-05-18:
I'm not sure if I understand what you're saying. I'm not using HifiBerryOS, just default Raspbian (for now). I've installed the `dspdac-noautomute.xml` sample profile and was running the `sigmatcpserver` with the `--lgsoundsync` flag, but it didn't work. As far as I could figure it out it was because the `readSPDIFOnRegister` attribute is not defined in the profile I used.

I've tried adding `<metadata type="readSPDIFOnRegister">93</metadata>` to my profile, but that doesn't work. It is probably is the wrong value (as the value is being logged...) As you can tell I have no idea how those profiles work...

hifiberry, 2020-05-18:
The profile you're using doesn't support the readSPDIFOnRegister attribute. Just adding some metadata record won't work as the profile itself doesn't support it.

I linked the other profile above to chow you a profile that does have support for the readSPDIFOnRegister attribute.

thomasjonas, 2020-05-18:
Aaah, now I understand it. Thanks. Unfortunately SigmaStudio is not available for macos. Although it's probably to complicated for me anyways... I might the profile you linked to. I probably don't need any special stuff anyways. Thanks again! And thanks for providing a nice product (especially the open-source software!).

closed 2021-02-16 by hifiberry