hifiberry/hifiberry-dsp #17: If the volume is set very low, the DAC+ DSP mutes

PauLL54, 2020-06-19:
If the volume is set very low, the DSP mutes for a short moment and the green LED is off.
I installed the dsptoolkit and the default profile.

Is this a software configuration issue or a hardware issue?

hifiberry, 2020-06-22:
This is neither a configuration nor a hardware issue, but a feature of the DSP profile. It's supposed to mute if no sound is detected. You can use another DSP profile without the auto-mute functionality, e.g. this one:
https://github.com/hifiberry/hifiberry-dsp/blob/master/sample_files/xml/dspdac-noautomute.xml

closed 2020-06-25 by hifiberry

PauLL54, 2020-06-26:
Thank you for your answer.
Would it be an idea to mention this feature in the documentation?
I really thought that I had damaged the board.
By the way, today I succeeded in creating a room correction file.
It was very easy and the result is astonishing!
I documented my findings here: https://www.scratchart.nl/mediaserver