hifiberry/hifiberry-dsp #22: Required on every boot

georgefst, 2020-10-24:
Since I first used this tool, I've found that I need to set it up again (e.g. run `dsptoolkit install-profile https://raw.githubusercontent.com/hifiberry/hifiberry-dsp/master/sample_files/xml/dacdsp-default.xml`) on every reboot in order to get any audio output from my DAC+ DSP.

This isn't too much of an inconvenience as I rarely reboot my Pi, but I'm just worried about whether it's symptomatic of a more serious problem.

Pi 4, Raspbian Buster, Linux `4.19.97-v7l+`

marecabo, 2021-02-02:
I am also experiencing that same issue.

In addition, after installing a profile, the `store`/`store-filters` commands yield this error (not sure if related):

```python
Traceback (most recent call last):
  File "/usr/local/bin/dsptoolkit", line 29, in <module>
    cmdline.main()
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 1019, in main
    self.command_map[cmd]()
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 882, in cmd_store
    self.store_attributes()
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/dsptoolkit.py", line 957, in store_attributes
    xmlprofile.read_from_text(xml.decode("utf-8", errors="replace"))
  File "/usr/local/lib/python3.7/dist-packages/hifiberrydsp/parser/xmlprofile.py", line 142, in read_from_text
    self.doc = xmltodict.parse(xmlcontent)
  File "/usr/local/lib/python3.7/dist-packages/xmltodict.py", line 327, in parse
    parser.Parse(xml_input, True)
xml.parsers.expat.ExpatError: syntax error: line 1, column 0
```

**System**

RPi4B
hifiberrydsp: 0.20
Raspian 10 (buster) / Linux moode 5.4.77-v7l+ #1371 SMP Tue Nov 17 13:35:27 GMT 2020 armv7l GNU/Linux

ziobombo, 2021-04-09:
I have exactly the same problem here: need to `dsptoolkit install-profile` at each reboot.
When trying to `store` and `store-filters`, I get the same stack trace as marecabo.

thomasschalch, 2021-04-17:
I've just received my first hifiberry DAC+ DSP board and I came across the same issue. So I'm not an expert at all, but maybe this helps you:
 * The DAC+ DSP board seems to have a **selfboot jumper** located right behind the TOSLINK input. As far as I understand, if this selfboot jumper is set, the DSP program is loaded from the persistent storage (EEPROM) right after powering on. Setting this jumper solved the issue for me.
(source: https://support.hifiberry.com/hc/en-us/search?utf8=%E2%9C%93&query=selfboot&commit=Search )
 * The dsptoolkit command `install-profle` writes the DSP program to the EEPROM **and** loads it on the DSP.
 (source: https://github.com/hifiberry/hifiberry-dsp/blob/master/doc/dsptoolkit.md )

It would be great, if an expert ( @hifiberry ?) could confirm my statements. It would be very helpful, if this jumper would also be mentioned in the datasheet of the DAC+ DSP ( https://www.hifiberry.com/docs/data-sheets/datasheet-dac-dsp/ ). The jumper on my DAC+ DSP board was not set when buying it. Is this really the intended default setting?

chatziko, 2021-12-05:
@thomasschalch : thanks! This drove me crazy until I found your post :smile: 
I wonder whether they just forgot to put the jumber at the factory...

georgefst, 2021-12-06:
Thanks @thomasschalch, that worked! In particular, inspired by [this forum post](https://support.hifiberry.com/hc/en-us/community/posts/360006076085-Can-t-program-DSP-to-selfboot), this fixed things permanently for me:

- power off the Pi
- remove the jumper
- power on
- run `dsptoolkit install-profile dacdsp-default.xml`
- power off 
- replace the jumper
- power on

georgefst, 2021-12-06:
And, of course, this really ought to be better documented.

chatziko, 2021-12-06:
Mine had no jumper at all, thankfully I had one lying around :smile:

DanGTG, 2022-04-29:
The steps above will work, however, in my case using a freshly imaged HiFiBerryOS I had to use the url in the first post instead of the non-existing local .xml.

georgefst, 2022-09-02:
Anyway, this is solved (although I'd reiterate that the jumper should be mentioned in documentation somewhere).

closed 2022-09-02 by georgefst