hifiberry/hifiberry-dsp #31: alsasync.py with pyalsaaudio 0.8+

mesufesu, 2021-03-09:
I am using pyalsaaudio 0.9.0 and the --alsa part fails to get DSPVolume control.
According to the documentation, in 0.8 mixer routines introduced device argument, and if its (or cardid) not provided the routines fall back to 'default' card which is not our hifiberry board.
If the device name is the same across all hifiberry dsp boards then its easy to hardcode it.
See https://github.com/mesufesu/hifiberry-dsp/commit/7cc2acd905aa65033d8da60085899605ce1e6559

Additionally, I think there is an indent mistake, because the code should not fail when we successfully create and obtain the mixer (https://github.com/mesufesu/hifiberry-dsp/commit/417f1996b0569868d23fa9ceaba48821954399d0)

ssalonen, 2023-03-02:
Is this fixed in https://github.com/hifiberry/hifiberry-dsp/commit/bb3c7c68f2f1f0f29ff982f3bb74d60e301140aa ? @mesufesu

hifiberry, 2025-06-11:
The latest version is compatible with pyaslaaudio 0.8 and 0.9

closed 2025-06-11 by hifiberry