hifiberry/hifiberry-dsp #33: dacdsp-v12.dspproj missing in this repo

marek-obuchowicz, 2021-07-03:
hifiberry-os currently includes updated dacdsp profile, version 12. This repository contains only the older version, v11, in `sample_files/dspproj` directory. Can we expect SigmaStudio profile version 12 to be added to this repo?

Reason for asking: I'm having trouble with TosLink In in v11 (both pre-compiled profile from hifiberry-os as well as source profile uploaded w/ SigmaStudio) - input detection block does not recognise incoming signal, even though there is a SPDIF-Lock and there is signal coming in, with severe volume). It works fine with dacdsp-v12.xml (included in hifiberry-os). Therefore, uploading a fixed project v12 here would be highly appreciated.

jarvick257, 2021-07-10:
I was facing the exact issue. After finding out about v12 (I'm not using HifiBerryOs so that took a while) I checked the diff between v11 and v12 xmls and noticed the asmrc input has changed [from 2 to 3](https://imgur.com/a/asWWMKw).
I then looked for this parameter in SigmaStudio and found it in the RegisterControls > ROUTING_MATRIX tab. ASRC7, which is supposed to handle SPDIF input, had its source set to "From DSP core outputs". After changing this to "From SPDIF receiver" toslink input worked perfectly!

[This is what it should look like.](https://imgur.com/a/fPlU3mz)

I still wish HiFiBerry would upload the v12 project here, though.

marek-obuchowicz, 2021-07-14:
@jarvick257 thanks for your response, that's actually a great way to reverse-engineer changes. Is there anything else that differs between those two profiles, or just this ASRC routing change?

jarvick257, 2021-07-15:
Yes there were a couple more changes but most of those look pretty cryptic in the xml so not really sure what they are.

marek-obuchowicz, 2021-07-15:
I asked because after changing ASRC input, I'm still getting issues with signal detection and mixing. I can workaround them in SigmaStudio by editing profile and adding a mixer instead of input detection logic (which seems to be broken). @hifiberry is there any chance to get any feedback from you? :)

hifiberry, 2021-07-15:
If you create push your own project via SigmaStudio you need to add the correct metadata to it. Make sure all the SPDIF registers are set correctly in your metadata. Read the registers and check iff all contain the correct values. 
Don't expect with ANY SigmaStudio project that you just push it to the DSP and it'll work. Without the correct metadata nothing will happen.

marek-obuchowicz, 2021-07-15:
sure, metadata is in place as described on your page. TosLink signal is detected correctly ("sources" website shows if / if not there is an incoming signal, it works ok) but still there's nothing on output after changing *just* the ASRC source - somehow v11 logic seems to be broken in terms of signal detection or mixing.

any chance for uploading sigmastudio project v12 to this repo?  Seems that the issue we're discussing here has already been solved by you :)

hifiberry, 2021-07-15:
There were no major changes in v12. The v12 project was never saved. Therefore, I can't upload it.

RPhilbo, 2021-09-21:
But there were changes that fixed problems, so it would help people to start with a sigmastudio project file v12 instead of v11, even the change is minor.
As marek said, the problem is already solved by you, so it would be helpful to not research it again and again :-)

marek-obuchowicz, 2021-09-21:
@hifiberry maybe the fixed project file could be uploaded with another version name (like v12-1)? By doing that, we are not going to introduce any possible incompatiblity between already released v12 (where source project is missing), in case that there were indeed more changes to v12. That could unblock a few users, like those participating in this thread. If it helps, let me know, I can create a PR when there's some spare time.

RPhilbo, 2021-09-21:
In the community (forum) of Hifiberry this is also questioned from different people.
As far as i understand this following links answer of support team from today, the difference in v11 and v12 is not in the projectfile, it is only in the metadata. So this would be a more detailed explanation for the non existing v12 project file :-)

https://support.hifiberry.com/hc/en-us/community/posts/360014328577

RPhilbo, 2021-09-21:
> As far as i understand this following links answer of support team from today, the difference in v11 and v12 is not in the projectfile,

Ok, thats not true. 
I started with the SigmaStudio File v11 and had the above already mentioned problems with TOSLINK input. Spotify for example was no problem, but TOSLINK input was not detected nor played.

When i used the Hifiberry OS official v12 (and reboot) it worked.

Then i took the SigmaStudio project v11 and just changed the already above mentioned ASRC7 input from SPDIF, then compile, metadata and so on (and reboot) it works now with TOSLINK.

pinballpower, 2021-09-21:
Not sure what metadata you used on the v11 profile. However, feel free to create a pull request with your new file.

RPhilbo, 2021-09-24:
After my posting above i also had some issues with signal detection on TOSLINK. TOSLINK was playing, but the detection was not working, perhaps like @marek-obuchowicz also observed.

My attention went then to the "SPDIF on" switch in the "input detection" schematic. It was off, but i think it needs to be on. After that, it worked.

So i opened my first pull request with these changes:

- changed ASRC7 source to "From S/PDIF receiver"

- Schematic "Input Selection" Switch "SPDIF On" was off in v11, so i switched it to "on"

@marek-obuchowicz can you confirm this v12-1 with open and deploy, if this config will also work for you?

-------------------------------------------------------------------------------------------------------
For an possible v13 i also have found more settings i changed, testings are ongoing:

- Visual rearrangement for better readability and understanding
	- rearranged some blocks and data lines
	- rearranged order of schematic sheets, to be in order like appearance in the main sheet.

- Disabled not needed functions (less power consumption and heat dissipation)
	- "Power_Clocking":
		- Serial Input 3 disabled
		- Serial Input 1 disabled

		- PDM1 Input disabled
		- PDM0 Input disabled

		- ADC0 disabled

- SPDIF_RX "Output strength on lock loss" changed to weak, because seems "safer" to me.

	From the ADAU1451 Datasheet:
	S/PDIF receiver behavior in the event that lock is lost. Applies to the output of the recovered frame clock from the S/PDIF receiver.
		Strong; output is continued as well as is possible when the receiver notices a loss of lock condition, which may result in some data corruption.
		Weak; 	output is interrupted as soon as receiver notices a loss of lock condition. 

- GPIO13/MP13 used in schematic "Hardware mute"
	- not investigated yet, but config of MP13 seems to be input instead of output?

hifiberry, 2021-09-27:
@RPhilbo Thanks for this. About the changes you plan for a future version:

> Error or different point of view regarding channel assignment
It's not needed to change the DSP program for this, it can be directly changed in the HiFiBerryOS GUI. 

>Also a little odd: i did the routine with programming, metadata etc. a douzen times, but the checksum just changed one time, instead for every little detail changed.
Checksum is only calculated on the program, NOT the register set. Therefore, changing some register settings won't change the checksum.

marek-obuchowicz, 2021-09-29:
@RPhilbo I just ordered toslink cables, have none free at the moment to connect hifiberry. As soon as they arrive (i'd expect next week), i'll be able to check your profile. Is there an XML profile, built from your project, which I could check, or do you prefer me to go with the sigmastudio build process and add metadata?

RPhilbo, 2021-09-30:
@marek-obuchowicz of course i can provide the xml file, maybe i also do a PR to put it into the xml folder.