hifiberry/hifiberry-dsp #35: Parameter control not working

elagil, 2021-09-11:
Hello,

I followed the documentation on adding parameters by means of `mergeparameters` the to my DSP profile .xml - this shall allow me to use volume control.

For this purpose, my application includes a volume control slider, which was labeled "Volume". Indeed, `mergeparameters` correctly detects the volume control register:

```
$ mergeparameters blus_XT25TG-30_22W_4534G00_v1.xml blus_XT25TG-30_22W_4534G00_v1.params
added parameters to XML profile:
  volumeControlRegister
```

and adds it to the .xml correctly as well:

```
<?xml version="1.0" encoding="utf-8"?>
<ROM IC="ADAU1452" IC_Address="1" Address_byte_length="2">
        <beometa>
                <metadata type="volumeControlRegister">272</metadata>
        </beometa>
        <dateTime>2021-09-11T11:11:15.0679920Z</dateTime>
        <version>4.6.0.1812</version>
        <page modetype="Mode 0">
                <action instr="writeXbytes" len="4" addr="62464" AddrIncr="0" ParamName="IC 2.HIBERNATE" DspName="IC 2">00 01</action>
                <action instr="delay" ParamName="IC 2.Hibernate Delay" DspName="IC 2">00 FF</action>
// more code ...
```

Then, I load this file to my DSP with 

```
dsptoolkit install-profile profile/blus_XT25TG-30_22W_4534G00_v1.xml
```

which works, and after 

```
dsptoolkit reset
```

the application is correctly loaded from EEPROM and the DSP runs as expected.

Now **the problem** is that the volume control parameter cannot be used, as it says

```
$ dsptoolkit get-volume
Profile doesn't support volume control
```

Having a look at the logs for sigmatcp:

```
$ sudo service sigmatcp status
● sigmatcp.service - SigmaTCP Server for HiFiBerry DSP
   Loaded: loaded (/lib/systemd/system/sigmatcp.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-09-11 12:28:24 BST; 8min ago
 Main PID: 623 (sigmatcpserver)
    Tasks: 1 (limit: 1438)
   CGroup: /system.slice/sigmatcp.service
           └─623 /usr/bin/python3 /usr/local/bin/sigmatcpserver --alsa

Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: INFO:root:writing XML file: b'<?xml version="1.0" encoding="utf-8"?>\n<ROM IC="ADAU1452" IC_Address="1" Address_byte_length="2">\n\t<beometa>\n\t\t<metadata type="volume
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: D C0 00 08 0F 8D 84 00 FC C0 00 08 0F 8D 94 01 01 C0 00 00 18 82 80 1D BE C0 00 00 18 82 80 9D BE 0C 41 FD A0 00 10 80 41 0D 40 01 01 00 40 20 08 C0 00 00 10 82 CC 1D DE
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: INFO:root:preparing for memory update
Sep 11 12:29:48 raspberrypi sigmatcpserver[623]: INFO:root:finished memory update
Sep 11 12:29:49 raspberrypi sigmatcpserver[623]: INFO:root:caching program memory checksum
Sep 11 12:29:49 raspberrypi sigmatcpserver[623]: INFO:root:reading profile $ sudo service sigmatcp status
● sigmatcp.service - SigmaTCP Server for HiFiBerry DSP
   Loaded: loaded (/lib/systemd/system/sigmatcp.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-09-11 12:28:24 BST; 8min ago
 Main PID: 623 (sigmatcpserver)
    Tasks: 1 (limit: 1438)
   CGroup: /system.slice/sigmatcp.service
           └─623 /usr/bin/python3 /usr/local/bin/sigmatcpserver --alsa

Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: INFO:root:writing XML file: b'<?xml version="1.0" encoding="utf-8"?>\n<ROM IC="ADAU1452" IC_Address="1" Address_byte_length="2">\n\t<beometa>\n\t\t<metadata type="volume
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: D C0 00 08 0F 8D 84 00 FC C0 00 08 0F 8D 94 01 01 C0 00 00 18 82 80 1D BE C0 00 00 18 82 80 9D BE 0C 41 FD A0 00 10 80 41 0D 40 01 01 00 40 20 08 C0 00 00 10 82 CC 1D DE
Sep 11 12:29:20 raspberrypi sigmatcpserver[623]: INFO:root:preparing for memory update
Sep 11 12:29:48 raspberrypi sigmatcpserver[623]: INFO:root:finished memory update
Sep 11 12:29:49 raspberrypi sigmatcpserver[623]: INFO:root:caching program memory checksum
Sep 11 12:29:49 raspberrypi sigmatcpserver[623]: INFO:root:reading profile /var/lib/hifiberry/dspprogram.xml
Sep 11 12:30:07 raspberrypi sigmatcpserver[623]: ERROR:root:can't get attribute volumeControlRegister from XML
Sep 11 12:30:07 raspberrypi sigmatcpserver[623]: ERROR:root:can't get attribute volumeControlRegister from XML
```

It cannot read the attribute from the XML. I checked the file `/var/lib/hifiberry/dspprogram.xml`, and it was correctly copied from my input .xml, so does contain the volumeControlRegister field.

What is the problem here?
Thanks in advance.

elagil, 2021-09-11:
I now simply added the checksum field and it suddenly works. Is the checksum field not optional, but actually mandatory?

hifiberry, 2022-10-24:
Checksum is require for mergeparameters

closed 2022-10-24 by hifiberry