hifiberry/hifiberry-dsp #39: Cannot find DSPVolume when hifiberry is not the default alsa device (eg when running pulseaudio)

chatziko, 2021-12-06:
`sigmatcpserver` tries to open `DSPVolume` in the default alsa device. If hifiberry is not the default device, it fails with
```
ERROR:root:ALSA mixer DSPVolume not found"
```
and the alsa mixer becomes ususable (it does not control the DSP volume). Tested on pi4 with DAC+DSP.

This happens for instance if pulseaudio is running (default in recent Raspberry Pi OS), since a pulseaudio compatibility wrapper is set as the default alsa device.

Correct behaviour: when multiple alsa deviecs are present, `sigmatcpserver` should detect which is the hifiberry one.

Workaround: pulseaudio's wrapper alsa device can be disabled by commenting out the contents of `/etc/alsa/conf.d/99-pulse.conf` (this however means that a program cannot use alsa to output sound via pulseaudio).

RobisLV, 2022-11-15:
Hi! I do not know if this is still relevant but I made a fix for this issue. I will submit PR and link to the fix today.

hifiberry, 2022-11-15:
We're happy to look into a PR for this.

hifiberry, 2022-11-17:
PR merged

closed 2022-11-17 by hifiberry

chatziko, 2022-11-17:
Seems to work fine, thanks!

PS. A pypi release might be useful, there seem to be a few changes since the last one.