hifiberry/hifiberry-dsp #48: beocreate v2.1 muted without "Selfboot" jumper

jedie, 2022-10-21:
I don't know if this is the right place or if it's a bug in HiFiBerryOS see https://github.com/hifiberry/hifiberry-os/issues/303#issuecomment-1285538403 or is https://github.com/bang-olufsen/create a better place?!?

I have a **beocreate** (On the PCB has `v2.1` printed on it) and run **HiFiBerryOS** v20220826

All channels are "muted" (No `MUTE` jumper is set). On the HiFiBerryOS web page, the DSP program was listed as "`Unknown Program`". I tried to install the "`Beocreate Universal`" program via web page, but it result always in "`Unknown Program`".

I also tied `dsptoolkit install-profile /opt/beocreate/beo-dsp-programs/beocreate-universal-10.xml` and create a `/etc/hifiberry/config.fixed` without success. 

Work-a-round is to set the "Selfboot" jumper. After a reboot it worked (without a `/etc/hifiberry/config.fixed`)

Some information (with the working `Selfboot` jumper):
```
_______________________________________________________________
++ /opt/hifiberry/bin/check-system
HiFiBerryOS version: 20220826
Pi type: 3B
Pi model: Raspberry Pi 3 Model B Rev 1.2
Compatibility: No known problems
Sound card: Beocreate 4-Channel Amplifier
Sound card ID: 0
aplay 1: card 0 sndrpihifiberry [snd_rpi_hifiberry_dac]
aplay 2:  device 0 HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
Mixer: DSPVolume
Features: pi pi3orlater arm7 dsp localui
Power controller: Error: Read failed
Error: Read failed
not detected
Linux version: Linux beocreate 5.15.56-v7 #1 SMP Thu Aug 25 10:09:45 UTC 2022 armv7l GNU/Linux
Date and time: Fri Oct 21 11:11:23 CEST 2022
eth0: 192.168.6.96
eth0: 192.168.6.52
/data mounted: OK
/dev/dri/card0: OK
BT devices: 
WiFi devices: wlan0
audiocontrol2: running
beocreate2: running
bluetooth: not running
bluealsa: running
bluealsa-aplay: running
dlnampris: not running
mpd: running
pigpio: running
raat: not running
shairport-sync: not running
sigmatcp: running
snapcastmpris: not running
spotify: running
squeezelite: not running
sshd: running
ympd: not running
weston: not running
cog: not running
_______________________________________________________________
++ /opt/hifiberry/bin/detect-hifiberry
Mounting /boot
Detecting using I2C
Could not find an I2C enabled sound card
Assuming HiFiBerry DAC+ Light, Zero
_______________________________________________________________
++ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
_______________________________________________________________
++ /opt/hifiberry/bin/readhat
no vendor:no product
```

hifiberry, 2022-10-21:
It's not a "workaround", but expected behavior. The selfboot jumper must be set. Otherwise the DSP isn't starting. Removing the selfboot jumper is only needed for specific development tasks (when creating your own DSP profile using SigmaStudio)

closed 2022-10-21 by hifiberry

jedie, 2022-10-21:
Oh, wow. But why is this Jumper not set by factory?

Any why is it's not documented? e.g. here: https://www.hifiberry.com/beocreate/beocreate-doc/beocreate-first-steps/

Any i think it's a good idea to make this Text part **bold** on https://www.hifiberry.com/beocreate/beocreate-doc/beocreate-hardware-documentation/

[image]