hifiberry/hifiberry-dsp #53: mergeparameters AttributeError: 'dict' object has no attribute 'move_to_end'

corrpel, 2022-12-27:
Hello,
I am trying to use the merge tool for metadata, but i receive the following errors:

'''
pi@raspberrypi:~/phat $ mergeparameters dsc_phat.xml dsc_phat.params 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/hifiberrydsp/parser/xmlprofile.py", line 253, in update_metadata
    beometa = self.doc["ROM"]["beometa"]
KeyError: 'beometa'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mergeparameters", line 28, in <module>
    merge_params_main()
  File "/usr/local/lib/python3.9/dist-packages/hifiberrydsp/parser/sigmaparams.py", line 236, in merge_params_main
    params = pf.merge_params_into_xml(xmlfile)
  File "/usr/local/lib/python3.9/dist-packages/hifiberrydsp/parser/sigmaparams.py", line 190, in merge_params_into_xml
    xml.update_metadata(param_list)
  File "/usr/local/lib/python3.9/dist-packages/hifiberrydsp/parser/xmlprofile.py", line 257, in update_metadata
    self.doc["ROM"].move_to_end('beometa', last=False)
AttributeError: 'dict' object has no attribute 'move_to_end'
'''

Running python3.9 and and latest dsptoolkit from pip

KruzFX, 2023-01-11:
I just experienced the same problem. Two years ago when I last used the tool there was no such a problem. Maybe there is a problem with new python versions?

ssalonen, 2023-02-26:
Just hit this issue myself, it is due to backwards compatible change in xmltodict v0.13 (xmltodict is dsptoolkit dependency)

Fix is simple: xmltodict.parse() needs additional parameter dict_constructor=OrderedDict

KruzFX, 2023-03-01:
Thanks for the reply. But where is the file when installed on my Raspberry Pi?

ssalonen, 2023-03-02:
PR submitted in https://github.com/hifiberry/hifiberry-dsp/pull/54

commit 13ca4c85 (2023-03-02): Merge pull request #54 from ssalonen/xmltodict-fix

Fix for xmltodict v0.13. Fixes #53
commit 1c0b4648 (2023-03-02): Fix for xmltodict v0.13. Fixes #53

xmltodict v0.13 started using dict (instead of previous default
OrderedDict) when parsing xml with python < 3.7. See
https://github.com/martinblech/xmltodict/commit/231b4d4ef4e804ca0311182e8ec8874af98a1bb7

This restores the previous behaviour, using OrderedDict, making the
behaviour compatible with dsptoolkit. The change should be backwards
compatible with older versions of xmltodict, as dict_constructor
parameter has been there for quite some time.
closed 2023-03-02 by hifiberry