hifiberry/hifiberry-dsp #63: Wrong PROGRAM_END_SIGNATURE?

koalo, 2024-04-13:
Hi,
I just stumbled upon the PROGRAM_END_SIGNATURE in https://github.com/hifiberry/hifiberry-dsp/blob/bf432eb2695f92ff81d2c02e863734392481894b/hifiberrydsp/hardware/adau145x.py#L53 which is

02 C**2** 00 00 00 00 00 00 00 00 00 00

but the last bytes in https://github.com/hifiberry/hifiberry-dsp/blob/bf432eb2695f92ff81d2c02e863734392481894b/sample_files/xml/beocreate-default.xml#L123 are

02 C**0** 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

What is the reason for that? Is this a bug or a feature?

Thanks,
Florian

koalo, 2024-04-22:
Oh I see, my mistake. It is the correct one for
https://github.com/hifiberry/hifiberry-dsp/blob/bf432eb2695f92ff81d2c02e863734392481894b/sample_files/xml/beocreate-default.xml#L300

What is the background? Why are two different programs written in succession?

hifiberry, 2024-04-22:
> Why are two different programs written in succession?

A DSP profile basically consists of 2 DSP programs. One that write something to the DSP - in the "real" program and then the real program afterwards. This is due to the architecture of the DSP that requires a program to write the EEPROM.

koalo, 2024-04-22:
That makes sense, thanks!

closed 2024-04-22 by koalo