hifiberry/hifiberry-dsp #64: Please include latest beocreate-universal-11.xml in sample_files/xml

jspricke, 2024-08-14:
Hi,

I am working on packaging this for Debian and would like to include a dsp program for the Beocreate. I think the one used in HifiberryOS is from here: https://github.com/hifiberry/create/tree/development/Beocreate2/beo-dsp-programs.

Could you also tag a new release once this is done?

hifiberry, 2024-08-21:
We won't include this here. This repository is not the source for the latest DSP profiles. 
DSP profiles are in this repository:
https://github.com/hifiberry/dspprofiles

I'll add the Beocreate profile to this repo soon as it's not yet included.

closed 2024-08-21 by hifiberry

jspricke, 2024-08-21:
Works for me. Would it make sense to remove the `sample_files` folder here?