hifiberry/hifiberry-dsp #67: Problem installing dsptoolkit on raspbian

Pantarhei-art, 2024-11-02:
I can't install dsptoolkit on my RPI 4 with Raspbian. I use the command: "sudo pip3 install --upgrade hifiberrydsp",
I have python v. 3.11 and PIP v. 23.0.1 installed, and yet I get this message:

```
"error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification."
```
I tried the method from the third paragraph above, but it doesn't work or I entered the wrong thing in "xyz":

hifiberry, 2024-12-16:
Debian doesn't allow system-wide installations of packages with pip anymore by default. Try the updated installation script.

hifiberry, 2025-05-11:
https://www.hifiberry.com/blog/package-repository-for-raspberry-pi-os/

closed 2025-05-11 by hifiberry