hifiberry/hifiberry-dsp #69: Samplerate not read correctly

rawdlite, 2024-12-08:
https://github.com/hifiberry/hifiberry-dsp/blob/3384c0cd26088a1c784942c9e23f1f5ca3cd25fb/hifiberrydsp/parser/xmlprofile.py#L292

reads xml attribute 'samplerate' yet its 'sampleRate' in provided profiles.
Also returning a valid value (48000) on error is confusing.
Returning 'sampleRate could not be read from profile metadata' instead would be more obvious.

rawdlite, 2024-12-09:
Just found in the code installed via pypi this being handled by a second Attribute self.sigmatcp.request_metadata(ATTRIBUTE_SAMPLERATE_CAP))
So it is working for me, but i am still confused.

hifiberry, 2024-12-16:
Unfortunately this is a historical thing as the samplerate attribute had been implemented as sampleRate and samplerate in different places.

48000 is the default when no sample rate is defined in the DSP profile. In this case, a default value is used.

closed 2024-12-16 by hifiberry