hifiberry/hifiberry-dsp #73: Use tonecontrol to implement loudness

rawdlite, 2024-12-16:
This is a suggestion:

When a profile does not support Loudness it could be to some degree achieved via tonecontrol.
I think of something like:

LOUDNESS = {
1:  {'hs': {'freq': '7000Hz', 'vol': '3dB'},
      'ls': {'freq': '100Hz', 'vol': '3dB'}},
2: {'hs': {'freq': '7000Hz', 'vol': '4dB'},
      'ls': {'freq': '150Hz', 'vol': '6dB'}},
3: {'hs': {'freq': '7000Hz', 'vol': '6dB'},
      'ls': {'freq': '200Hz', 'vol': '9dB'}}}

then in cmd_set_loudness if the profile does not support loudness call tonecontrol with these values.

**Note**: these are just exemplary values and should be adapted by someone more knowledgeable than me. I just had a brief look at the Fletcher-Munson curves.
https://en.wikipedia.org/wiki/Equal-loudness_contour

A dsp profile with adaptive loudness would of course be superior. Alas my knowledge of sigma studio is (yet) not up to that task. Until then this is a work around.

hifiberry, 2024-12-17:
While I understand the idea, dsptoolkit this would add a lot of complexity into dsptoolkit. This is definitely something that should be handled by an external tool (that could use dsptoolkit to add these filters).

rawdlite, 2024-12-17:
Fair enough.
I just saw someone complain about loudness not being supported by the profile. 
Yet I see your point.  What about a contrib dir in the repository where these kind of 'helper' programs could be hosted?
Another idea is a 'fake' mute program using set-volume.

hifiberry, 2024-12-18:
Sure, feel free to add something in a crontrib directory

closed 2024-12-18 by hifiberry